rosflight/rosflight_firmware #271: Not compiling on arm-none-eabi-gcc 6.3.1

korken89, 2017-11-30:
*1. OS Version*
Ubuntu 16.04

*2. git tag (master, stable, v1.0, etc..)*
master

*3. parameter file (`rosservice call /param_save_to_file`)*
None

*4. Compiler version (`arm-none-eabi-gcc --version`)*
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]

*5. Hardware configuration (MAV type, flight controller version, connected sensors)*
None

*Compile bug:*
When trying to compile the firmware I get the following error:

```
In file included from /usr/arm-none-eabi/include/stdlib.h:11:0,
                 from /usr/arm-none-eabi/include/c++/6.3.1/cstdlib:75,
                 from /usr/arm-none-eabi/include/c++/6.3.1/stdlib.h:36,
                 from lib/breezystm32/breezystm32.h:25,
                 from naze32.cpp:38:
lib/breezystm32/breezyprintf.h:118:16: error: conflicting declaration of C function 'int tfp_printf(const char*, ...)'
 #define printf tfp_printf 
                ^
In file included from lib/breezystm32/breezystm32.h:46:0,
                 from naze32.cpp:38:
lib/breezystm32/breezyprintf.h:113:6: note: previous declaration 'void tfp_printf(const char*, ...)'
 void tfp_printf(const char *fmt, ...);
      ^~~~~~~~~~
In file included from /usr/arm-none-eabi/include/stdlib.h:11:0,
                 from /usr/arm-none-eabi/include/c++/6.3.1/cstdlib:75,
                 from /usr/arm-none-eabi/include/c++/6.3.1/stdlib.h:36,
                 from lib/breezystm32/breezystm32.h:25,
                 from naze32.cpp:38:
lib/breezystm32/breezyprintf.h:119:17: error: conflicting declaration of C function 'int tfp_sprintf(char*, const char*, ...)'
 #define sprintf tfp_sprintf 
                 ^
In file included from lib/breezystm32/breezystm32.h:46:0,
                 from naze32.cpp:38:
lib/breezystm32/breezyprintf.h:114:6: note: previous declaration 'void tfp_sprintf(char*, const char*, ...)'
 void tfp_sprintf(char* s, const char *fmt, ...);
      ^~~~~~~~~~~
In file included from /usr/arm-none-eabi/include/c++/6.3.1/ext/string_conversions.h:43:0,
                 from /usr/arm-none-eabi/include/c++/6.3.1/bits/basic_string.h:5417,
                 from /usr/arm-none-eabi/include/c++/6.3.1/string:52,
                 from /usr/arm-none-eabi/include/c++/6.3.1/stdexcept:39,
                 from /usr/arm-none-eabi/include/c++/6.3.1/array:39,
                 from /usr/arm-none-eabi/include/c++/6.3.1/tuple:39,
                 from /usr/arm-none-eabi/include/c++/6.3.1/functional:55,
                 from ../../include/board.h:35,
                 from naze32.h:41,
                 from naze32.cpp:45:
/usr/arm-none-eabi/include/c++/6.3.1/cstdio:127:11: error: '::printf' has not been declared
   using ::printf;
           ^~~~~~
/usr/arm-none-eabi/include/c++/6.3.1/cstdio:137:11: error: '::sprintf' has not been declared
   using ::sprintf;
           ^~~~~~~
Makefile:218: recipe for target 'build/rosflight/./naze32.o' failed
make[1]: *** [build/rosflight/./naze32.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from /usr/arm-none-eabi/include/stdlib.h:11:0,
                 from /usr/arm-none-eabi/include/c++/6.3.1/cstdlib:75,
                 from /usr/arm-none-eabi/include/c++/6.3.1/stdlib.h:36,
                 from lib/breezystm32/breezystm32.h:25,
                 from naze32.h:38,
                 from main.cpp:32:
lib/breezystm32/breezyprintf.h:118:16: error: conflicting declaration of C function 'int tfp_printf(const char*, ...)'
 #define printf tfp_printf 
                ^
In file included from lib/breezystm32/breezystm32.h:46:0,
                 from naze32.h:38,
                 from main.cpp:32:
lib/breezystm32/breezyprintf.h:113:6: note: previous declaration 'void tfp_printf(const char*, ...)'
 void tfp_printf(const char *fmt, ...);
      ^~~~~~~~~~
In file included from /usr/arm-none-eabi/include/stdlib.h:11:0,
                 from /usr/arm-none-eabi/include/c++/6.3.1/cstdlib:75,
                 from /usr/arm-none-eabi/include/c++/6.3.1/stdlib.h:36,
                 from lib/breezystm32/breezystm32.h:25,
                 from naze32.h:38,
                 from main.cpp:32:
lib/breezystm32/breezyprintf.h:119:17: error: conflicting declaration of C function 'int tfp_sprintf(char*, const char*, ...)'
 #define sprintf tfp_sprintf 
                 ^
In file included from lib/breezystm32/breezystm32.h:46:0,
                 from naze32.h:38,
                 from main.cpp:32:
lib/breezystm32/breezyprintf.h:114:6: note: previous declaration 'void tfp_sprintf(char*, const char*, ...)'
 void tfp_sprintf(char* s, const char *fmt, ...);
      ^~~~~~~~~~~
In file included from /usr/arm-none-eabi/include/c++/6.3.1/ext/string_conversions.h:43:0,
                 from /usr/arm-none-eabi/include/c++/6.3.1/bits/basic_string.h:5417,
                 from /usr/arm-none-eabi/include/c++/6.3.1/string:52,
                 from /usr/arm-none-eabi/include/c++/6.3.1/stdexcept:39,
                 from /usr/arm-none-eabi/include/c++/6.3.1/array:39,
                 from /usr/arm-none-eabi/include/c++/6.3.1/tuple:39,
                 from /usr/arm-none-eabi/include/c++/6.3.1/functional:55,
                 from ../../include/board.h:35,
                 from naze32.h:41,
                 from main.cpp:32:
/usr/arm-none-eabi/include/c++/6.3.1/cstdio:127:11: error: '::printf' has not been declared
   using ::printf;
           ^~~~~~
/usr/arm-none-eabi/include/c++/6.3.1/cstdio:137:11: error: '::sprintf' has not been declared
   using ::sprintf;
           ^~~~~~~
Makefile:218: recipe for target 'build/rosflight/./main.o' failed
make[1]: *** [build/rosflight/./main.o] Error 1
rm lib/breezystm32/startup_stm32f10x_md_gcc.s
make[1]: Leaving directory '/home/emifre/Git/rosflight_firmware/boards/naze'
Makefile:54: recipe for target 'all' failed
make: *** [all] Error 2
```

superjax, 2017-11-30:
Thanks for documenting this, this is actually a known bug that we've pretty much been ignoring.  I'm not sure why gcc 6 has a different include structure than gcc 5 on the same Makefile.  It looks like it's including the breezySTM32 version of `nanoprintf` which is wrong, and is probably causing namespace issues with `cstdio`

len0rd, 2017-11-30:
@korken89 I'm also running gcc 6.3.1, and it looks like the feature_85 branch has partially resolved this issue, although it still has build errors:

1. _OS Version_
MacOS 10.13.1

2. git tag (master, stable, v1.0, etc..)
feature_85

3. parameter file (rosservice call /param_save_to_file)
None

4. Compiler version (arm-none-eabi-gcc --version)
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]

5. Hardware configuration (MAV type, flight controller version, connected sensors)
None

Compile bug:
When trying to compile the firmware I get the following error:

```
%% state_manager.cpp
../../src/param.cpp: In member function 'void rosflight_firmware::Params::add_callback(std::function<void(int)>, uint16_t)':
../../src/param.cpp:254:52: error: unused parameter 'callback' [-Werror=unused-parameter]
 void Params::add_callback(std::function<void(int)> callback, uint16_t param_id)
                                                    ^~~~~~~~
../../src/param.cpp:254:71: error: unused parameter 'param_id' [-Werror=unused-parameter]
 void Params::add_callback(std::function<void(int)> callback, uint16_t param_id)
                                                                       ^~~~~~~~
../../src/param.cpp: In member function 'void rosflight_firmware::Params::change_callback(uint16_t)':
../../src/param.cpp:290:39: error: unused parameter 'id' [-Werror=unused-parameter]
 void Params::change_callback(uint16_t id)
```

superjax, 2017-11-30:
I just pushed an update to the `fix_271_printf_issue` branch.  Could you give it a try? (be sure to update submodules) I just successfully compiled it with gcc-arm-none-eabi 6.3.1 (from the PPA).

I haven't run it though, I don't have hardware with me at the moment.

len0rd, 2017-11-30:
Resolves it for me.

korken89, 2017-11-30:
Yeah, the branch is compiling fine!

superjax, 2017-11-30:
Cool.  Let me do some hardware testing and I'll merge it into master

closed 2018-02-23 by superjax