rosflight/rosflight_firmware #276: SBUS Support

superjax, 2018-01-18:
It would be awesome if we could use FrSky receivers SBUS protocol.

superjax, 2018-01-18:
Actually, on second thought, this isn't really necessary if we support PPM

DaneSlattery, 2019-02-08:
discuss.rosflight.org seems to be broken, but rather than opening a new issue, I thought I would just comment here.

http://docs.rosflight.org/en/latest/user-guide/hardware-setup/ says only PPM is supported, but 
http://docs.rosflight.org/en/latest/user-guide/rc-configuration/ says ROSflight only supports PPM receivers on F1 targets while F4 receivers support SBUS and PPM.

Which one is true, and if it is the latter, should the docs reflect this?

superjax, 2019-02-10:
Thanks for pointing that out.  The latter is correct, F4 targets support sbus while f1 do not.  

Yeah, sorry about the forum being down.  We are working on a new hosting solution and we should be up again soon.

superjax, 2019-02-10:
I reopened  this to remind me to modify the did to not be confusing.

bsutherland333, 2023-10-04:
Documentation has already been clarified.

closed 2023-10-04 by bsutherland333