rosflight/rosflight_firmware #282: rc_raw returning all 0

shlapion, 2018-02-15:
Hello everybody, 

was wondering if you could point me in the right direction. 

I have followed the instructions listed in the wiki. However when i eco the rc_raw topic it seems like there is no signal being received form the RC. The receiver and the RC have been bound and the green light is on on the receiver.

My setup is as follows:

- Naze 32 Rev 6 running firmware V1.1.0

- Futaba T16SZ RC 

- Futaba R7008SB receiver 

- ROS Kinetic running on Ubuntu 16.04
 

Any help is much appreciated

dpkoch, 2018-02-15:
What protocol are you trying to use to connect the receiver to the flight controller (PPM, PWM, SBUS, etc.)?

shlapion, 2018-02-16:
Hello @dpkoch, 

after a few hours i realized i am using SBUS. 

I am currently searching how to change the configuration on my receiver to PPM. 

Thx for the info. 

Will there be any support for SBUS in the future?

superjax, 2018-02-21:
Probably not for F1 targets.  The F4 release (hopefully this month) will include SBUS support.

shlapion, 2018-02-21:
Thank you for the info.

closed 2018-02-21 by shlapion