rosflight/rosflight_firmware #293: Naze32 rev6, barometer seems to not work

korken89, 2018-05-14:
Hi, the reading of the barometer gives nothing, but it is coming at what seems to be a normal rate.
The message in ROS looks like this, but is stuck on the same value always:

```
header: 
  seq: 79283
  stamp: 
    secs: 1526290026
    nsecs: 174457882
  frame_id: ''
altitude: -0.023193359375
pressure: 85729.125
temperature: 0.0
```

Tell me if there is something I can do to help debug this issue!

1. Tested on Ubuntu 16.04, ROS Kinetic
2. The Naze32 is flashed with the file refereed to in the setup guide (v1.1), bought from Hobbyking so it is the "AfroFlight32" which should be exactly the same as the Naze32. 
3. parameter file (`rosservice call /param_save_to_file`)
4. Not applicable, firmware downloaded
5. Only the board, just testing sensors

Parameter file:

```
- {name: ACC_LPF_ALPHA, type: 9, value: 0.5}
- {name: ACC_X_BIAS, type: 9, value: 0.06633938848972321}
- {name: ACC_X_TEMP_COMP, type: 9, value: 0}
- {name: ACC_Y_BIAS, type: 9, value: 0}
- {name: ACC_Y_TEMP_COMP, type: 9, value: 0}
- {name: ACC_Z_BIAS, type: 9, value: 0.4218389093875885}
- {name: ACC_Z_TEMP_COMP, type: 9, value: 0}
- {name: AIL_REV, type: 6, value: 0}
- {name: ARM_CHANNEL, type: 6, value: -1}
- {name: ARM_SPIN_MOTORS, type: 6, value: 1}
- {name: ARM_THRESHOLD, type: 9, value: 0.1500000059604645}
- {name: BARO_BIAS, type: 9, value: -85729.125}
- {name: BAUD_RATE, type: 6, value: 921600}
- {name: CAL_GYRO_ARM, type: 6, value: 0}
- {name: DIFF_PRESS_BIAS, type: 9, value: 0}
- {name: ELEVATOR_REV, type: 6, value: 0}
- {name: FAILSAFE_THR, type: 9, value: 0.300000011920929}
- {name: FILTER_INIT_T, type: 6, value: 3000}
- {name: FILTER_KI, type: 9, value: 0.05000000074505806}
- {name: FILTER_KP, type: 9, value: 0.5}
- {name: FILTER_MAT_EXP, type: 6, value: 1}
- {name: FILTER_QUAD_INT, type: 6, value: 1}
- {name: FILTER_USE_ACC, type: 6, value: 1}
- {name: FIXED_WING, type: 6, value: 0}
- {name: GROUND_LEVEL, type: 9, value: 1387}
- {name: GYRO_LPF_ALPHA, type: 9, value: 0.300000011920929}
- {name: GYRO_X_BIAS, type: 9, value: 0.04019036889076233}
- {name: GYRO_Y_BIAS, type: 9, value: 0.01206468790769577}
- {name: GYRO_Z_BIAS, type: 9, value: 0.003864590311422944}
- {name: MAG_A11_COMP, type: 9, value: 1}
- {name: MAG_A12_COMP, type: 9, value: 0}
- {name: MAG_A13_COMP, type: 9, value: 0}
- {name: MAG_A21_COMP, type: 9, value: 0}
- {name: MAG_A22_COMP, type: 9, value: 1}
- {name: MAG_A23_COMP, type: 9, value: 0}
- {name: MAG_A31_COMP, type: 9, value: 0}
- {name: MAG_A32_COMP, type: 9, value: 0}
- {name: MAG_A33_COMP, type: 9, value: 1}
- {name: MAG_X_BIAS, type: 9, value: 0}
- {name: MAG_Y_BIAS, type: 9, value: 0}
- {name: MAG_Z_BIAS, type: 9, value: 0}
- {name: MIN_THROTTLE, type: 6, value: 1}
- {name: MIXER, type: 6, value: 2}
- {name: MOTOR_IDLE_THR, type: 9, value: 0.1000000014901161}
- {name: MOTOR_MAX_PWM, type: 6, value: 2000}
- {name: MOTOR_MIN_PWM, type: 6, value: 1000}
- {name: MOTOR_PWM_UPDATE, type: 6, value: 490}
- {name: OVRD_LAG_TIME, type: 6, value: 1000}
- {name: PARAM_MAX_CMD, type: 9, value: 1}
- {name: PID_PITCH_ANG_D, type: 9, value: 0.05000000074505806}
- {name: PID_PITCH_ANG_I, type: 9, value: 0}
- {name: PID_PITCH_ANG_P, type: 9, value: 0.1500000059604645}
- {name: PID_PITCH_RATE_D, type: 9, value: 0}
- {name: PID_PITCH_RATE_I, type: 9, value: 0}
- {name: PID_PITCH_RATE_P, type: 9, value: 0.07000000029802322}
- {name: PID_ROLL_ANG_D, type: 9, value: 0.05000000074505806}
- {name: PID_ROLL_ANG_I, type: 9, value: 0}
- {name: PID_ROLL_ANG_P, type: 9, value: 0.1500000059604645}
- {name: PID_ROLL_RATE_D, type: 9, value: 0}
- {name: PID_ROLL_RATE_I, type: 9, value: 0}
- {name: PID_ROLL_RATE_P, type: 9, value: 0.07000000029802322}
- {name: PID_TAU, type: 9, value: 0.05000000074505806}
- {name: PID_YAW_RATE_D, type: 9, value: 0}
- {name: PID_YAW_RATE_I, type: 9, value: 0}
- {name: PID_YAW_RATE_P, type: 9, value: 0.25}
- {name: RC_ATT_CTRL_CHN, type: 6, value: -1}
- {name: RC_ATT_MODE, type: 6, value: 1}
- {name: RC_ATT_OVRD_CHN, type: 6, value: 4}
- {name: RC_F_CHN, type: 6, value: 2}
- {name: RC_MAX_PITCH, type: 9, value: 0.7860000133514404}
- {name: RC_MAX_PITCHRATE, type: 9, value: 3.141590118408203}
- {name: RC_MAX_ROLL, type: 9, value: 0.7860000133514404}
- {name: RC_MAX_ROLLRATE, type: 9, value: 3.141590118408203}
- {name: RC_MAX_YAWRATE, type: 9, value: 1.506999969482422}
- {name: RC_NUM_CHN, type: 6, value: 6}
- {name: RC_OVRD_DEV, type: 9, value: 0.1000000014901161}
- {name: RC_THR_OVRD_CHN, type: 6, value: 4}
- {name: RC_TYPE, type: 6, value: 1}
- {name: RC_X_CHN, type: 6, value: 0}
- {name: RC_Y_CHN, type: 6, value: 1}
- {name: RC_Z_CHN, type: 6, value: 3}
- {name: RUDDER_REV, type: 6, value: 0}
- {name: STRM_AIRSPEED, type: 6, value: 20}
- {name: STRM_ATTITUDE, type: 6, value: 200}
- {name: STRM_BARO, type: 6, value: 50}
- {name: STRM_HRTBT, type: 6, value: 1}
- {name: STRM_IMU, type: 6, value: 500}
- {name: STRM_MAG, type: 6, value: 50}
- {name: STRM_RC, type: 6, value: 50}
- {name: STRM_SERVO, type: 6, value: 50}
- {name: STRM_SONAR, type: 6, value: 40}
- {name: STRM_STATUS, type: 6, value: 10}
- {name: SWITCH_5_DIR, type: 6, value: 1}
- {name: SWITCH_6_DIR, type: 6, value: 1}
- {name: SWITCH_7_DIR, type: 6, value: 1}
- {name: SWITCH_8_DIR, type: 6, value: 1}
- {name: SYS_ID, type: 6, value: 1}
- {name: X_EQ_TORQUE, type: 9, value: 0}
- {name: Y_EQ_TORQUE, type: 9, value: 0}
- {name: Z_EQ_TORQUE, type: 9, value: 0}
```

superjax, 2018-05-14:
I've seen this bug before, and I thought that it had been fixed a while back. (I'm pretty sure that it's actually a problem with the breezystm32 drive layer) I'll probably have time to look at this this afternoon or tomorrow.  Thanks for letting me know!

korken89, 2018-05-14:
Thanks for having a look!

korken89, 2018-05-14:
Hi, I was using the latest release (which at the current time seems to be `v1.1.0` as you say.
Is is safe to use the `master` branch? I am generally a bit careful when not running stable releases. :)

superjax, 2018-05-14:
What are you using it for?

I'm flying `master` for my own Ph.D. research efforts, and consider that use case very stable. I've also got a pull request open to release current `master` -  The differences are pretty small, but the one that you want is the BMP280 support.  However, I pretty much only fly multirotors.

superjax, 2018-05-15:
Just FYI, we just released `master`, so you should be able to use the new `.hex` file.

korken89, 2018-05-15:
Hi, thanks for the release and prompt reply!

I have my own FCU I have been supporting for about 5 years now, which was first my hobby and then it became the FCU I,and my group, used during my PhD. While this was fine, now when I'm finishing I want to leave the group with an FCU that I don't need to support. :)

So I have been testing Pixhawk, but was not very impressed. Too bloated, non deterministic communication, and the documentation was meh.
Now I'm testing rosflight, and so far it seems quite straight forward - which I like in its simplicity.
A bit lacking support for new motor protocols, ESC telemetry and serial receivers, but these are not "must have" features. 
However, from a research perspective, ESC telemetry and synchronous motor protocols are something I would recommend, together with tracking notch filters for the gyros to not be as sensitive to vibrations (and to practically remove low pass filters). :)
But if you are interested in something like that we can discuss it somewhere else, I have reference implementations for it all that I could port the day you use an F4 or better MCU.

Thanks again, and keep up the good work!

superjax, 2018-05-15:
No problem! 

Our F4 port is basically done (I'm running it right now on a final overnight test before merging into master) I've been flying the F4 for a couple of weeks.  

I think we would be super interested in those additions.  It makes a lot of sense to do notch filtering, and ESC telemetry would be really awesome.  Our loop time on the F4 is on average about 40 us, so there is actually a ton of processor overhead.  I'll start some issues for those topics and we can chat about them some more.  Thanks also for the feedback about motor mixing!

korken89, 2018-05-15:
That's really good news! I saw a bit about F4 here and there, but I did not have time to dig into it yet.
Let's continue where it fits, I also saw you had a discussion forum. I have a small writeup on recommended design practices for these type of systems which I can paste in, just for reference and discussion.

And no problem on motor mixing, you are having the same problem I had some time ago.
Just that I chose to solve it with a larger hammer and tested implementing an MPC, which an F7 can do, but I would not recommend.
The simplicity of the cascaded PID loop and some smart saturation really gets you close to all the way! :)

korken89, 2018-05-22:
Hi again, I should probably report that in v1.2 the barometer is working again! Thanks!

superjax, 2018-05-22:
Awesome

closed 2018-05-22 by superjax