rosflight/rosflight_firmware #295: ESC telemetry

superjax, 2018-05-15:
I've been out of the FPV game for a while, but @korken89 pointed out that the new ESC's have telemetry data.  I think it makes a ton of sense to have ESC telemetry on a research autopilot.

This is a two-part issue, because we would first need to develop hardware support for the ESC in airbourne and also put in the plumbing for handling the data and passing it out over the commlink.

I'm not quite sure how the merging of UARTs work,  I mean, look at this picture

[image]
(credit: oscar liang)

I find it strange that all the ESCs are able to join together onto a single UART Rx.  Are they listening for space to talk, like Wifi?  I need to do a bit more reading into this, perhaps @korken89 has insights here.

Shouldn't be too bad, but would be awesome.  @jerelbn, @dpkoch and I have been talking a lot about trying to feed-forward motor torques and forces for improved state estimation in wind, this could potentially be a source of information for these experiments.

korken89, 2018-05-15:
I have a lot of experience with the ESC telemetry and requirements around it, I can give it a writeup a little later today!

superjax, 2018-05-15:
Awesome!  Thanks!

korken89, 2018-05-15:
Hi, here is the write up, if you have any questions just ask :) I added explanation for both issues.
https://docs.google.com/document/d/1d1K9bWUar4ZtDkrcEYbvmIOPkQiSRlk5cxfb58fSTM8/edit?usp=sharing

superjax, 2018-05-16:
This is cool!

Okay, I don't have a ton of experience with dma timers.  We have dma drivers for pretty much all our peripherals, but I've never seen the need for dma on the timer.  How does that work?  Does the dma load the ccr values that create the whole message at once, or do you have to interrupt in all the time to basical

korken89, 2018-05-16:
For the timers, it works through a "trick" lets say, and I will explain it.

Overall, the `TIM_UP` event is generated whenever the TIMer reads the current compare value when in `PWM` mode, then `DMA` is triggered by this event to write a new compare value before the current `PWM` period is done (which is difficult to achieve when trying to do with CPU).
This allows one to change the width of the `PWM` in every cycle, which can be used to encode the "1" and "0" periods, or any other period for that matter - this is why by supporting one protocol, `DMA` based, one supports all. :)
The trick is that when one does not want a continuous `PWM` waveform after the end of a "packet" (DShot/Oneshot) one writes the last update as 0, disabling the `PWM`, and then for each packet the procedure is repeated.
It is also possible to disable the timer output instead of writing a 0.

So to summarize the functionality:
1. Set up `TIMx_UP` to trigger a `DMA` transaction
2. Create the buffer of timer compare values
3. Set the addresses and size for the buffer in the `DMA`
4. Start `DMA`, and the packet will be sent without any CPU interaction

When it comes to supporting Oneshot - this is up to you. In my system I supported it all just not have people ask "what about X protocol", as all are supported via the same `DMA` system, just different constants.

About using 490 Hz `PWM`, I would very strongly discourage  it.
While it can, and a lot of the time is, working fine the rolling time delay that is introduced causes havoc if you want to do estimation based on control signal (eg. online system identification). Plus, it's not an "elegant" solution ;)
I fought a lot with this in my own system, when working with online adaptive estimation/control until I found it out, but this was on a small and quite nimble system (~600g).
Remember that the worst time delay is 2x the PWM period which is about ~2ms.

bsutherland333, 2023-10-04:
I think this would still be very useful, so we may revisit this issue.

len0rd, 2023-10-04:
hey i worked on this for a bit when i was in the lab. From what i remember, the layout of TIM peripheral channels on the RevoF4 servo ports was not super-conducive to supporting 4-motor+ DSHOT. I would have to go back and look for all the details, but I think there was some kind of issue around using the TIMx_UP signal and DMA channel mapping. just a warning if the revo is still the primary hardware

bsutherland333, 2023-10-04:
Ok, that's good to know. The Revo isn't the primary board anymore, we've switched to using a custom H7 based flight controller with plans to add support for the CubePilot Orange as well. If ESC telemetry requires dshot then that may be more work than we want to do right now.

closed 2023-10-04 by bsutherland333