rosflight/rosflight_firmware #305: F4 Airspeed not working

sethholsinger, 2018-09-05:
I have been trying to get a openpilot revolution board to register a pixhawk airspeed sensor(JDrones). I have tried two boards and two i2c sensors both times the sensor did not register. I am using the flexiport (as it is the only one that seems to support i2c). Attached is my parameter file. Any ideas as to why this might be happening?

[F4_airspeed.txt](https://github.com/rosflight/firmware/files/2353522/F4_airspeed.txt)

superjax, 2018-09-05:
That is interesting.

One idea I have is to make sure that the board is powered with external 5V.

@BillThePlatypus, have you observed this?

BillThePlatypus, 2018-09-05:
I have not. I can check on it later today.

BillThePlatypus, 2018-09-05:
The power comment is really important. The usb port doesn't power the I2C.

sethholsinger, 2018-09-05:
I tried with and without power. Doesn't seem to register either way.

BillThePlatypus, 2018-09-05:
When you turn it on, do you leave the autopilot disarmed for a short time? It only looks for new sensors while the autopilot is disarmed.

sethholsinger, 2018-09-06:
I do not arm the autopilot as and just trying to test sensors.

sethholsinger, 2018-09-06:
Today a remade the cable that I am using to connect the i2c sensor to flexiport on the openpilot revo. And tried everything again and I am still not getting any airspeed data.

superjax, 2018-09-06:
That is kinda weird.  It is working on our end.

Could I recommend running the `airbourne` airspeed example?

`cd boards/airbourne/airbourne/examples/i2c_airspeed`
`make`
`make flash`
`miniterm.py /dev/ttyACM0`

You should be able to see the airspeed readings, and the yellow light (if you are using a revo) will go off.

superjax, 2018-09-06:
I also just tried it on the current `master` of `rosflight` and `master` on the `firmware`.  here is my setup.

[image]

dpkoch, 2018-09-06:
Have we verified that we're using the same sensor? The JDrones module uses a Measurement Specialties 4525DO, is that what we have? Also would it be worthwhile to run the `i2c_sniff` example from `airbourne_f4` to verify that the JDrones sensor shows up with the expected I2C address?

sethholsinger, 2018-09-06:
I will try what you have suggest and I will using an older i2c sensor as well. The new one that I am using from JDrones appears to be different from their previous offerings (mainly in that it has less markings then before). I will also check to see if there is any indication that it is an updated sensor.

sethholsinger, 2018-09-07:
From your picture it looks like you are using the 1.1 version of the JDrones airspeed sensor. The newest version that JDrones sells is v1.4. I am pretty sure this is the issue, but I will test the v1.1 sensor I have to confirm.

superjax, 2018-09-07:
Sounds good.

I'm curious if the i2c address has been changed (I believe it is possible to change the address).  You might try running the `i2c_sniff` example in the `airbourne/examples` directory and post the output.

sethholsinger, 2018-09-10:
So I have tried with v1.1 and 1.4 of the JDrones i2c sensors. Using the master version of the firmware v1.1 is recognized as airspeed sensor but it refuses to update and just reads a airspeed of 0.0 and a temperature that is a very small number. Master doesn't seem to recognize v1.4 at all. When I run using the i2c_airspeed example both v1.1 and v1.4 are recognized and give correct pressure and temperature readings.

superjax, 2018-09-13:
This is really strange.  The one I am showing in my picture is a v1.2.  I'll look around my lab and see if I can find a v1.1 and a v1.4.

superjax, 2018-09-13:
I just noticed that they changed the pinout between the v1.1 and the v1.2.  (and I accidentally burned up a v1.1)  This is probably a stupid question, but have you ensured that you've got it wired correctly.

superjax, 2018-09-13:
Okay, I am thinking that perhaps the auto detection feature of I2C devices isn't working.  Could you perhaps try calling `rosservice call /reboot` and see if the airspeed is detected if it's powered on the first time the FC looks for it?

sethholsinger, 2018-09-14:
I tried the reboot command and the barometer always calibrated, but the airspeed sensor never did. To that point when the airspeed does calibrate is very infrequent. I am beginning to wonder if the problem is a power issue as the sensor never updates with a value that is different from the initial value. I don't know what I can be doing incorrectly to power the board though as I have connected a motor and speed control to the board with a battery connected to the speed controller. I have tried connecting the motor three pin to every single position. As for the wiring layout I am quite sure that I have that correct (though like you I also almost burned out a i2c sensor the first time I tried it). But after I was getting data from the i2c_airspeed example on both 1.1 and 1.4 I am sure it is correct now.

superjax, 2018-09-25:
Sorry @sethholsinger, I still haven't been able to replicate this problem.  Should we schedule a hangouts so I could watch your setup?

sethholsinger, 2018-09-26:
Sure, let me know what date and time works best for you.

len0rd, 2019-01-31:
@sethholsinger are you still having this issue? Did you find a solution?

I am able to reproduce the problem where the airspeed topic freezes on an initial value. This occurs when external 5V power is momentarily disconnected from the board. When power is regained, the sensor/rostopic never recover. The i2c_airspeed example works fine and is able to recover from power interruptions.

Unfortunately I only have a v1.1 and 1.2 sensor so I'm unable to confirm any of the v1.4 issues.

I'm looking into a fix for this, but it may be worth verifying that your secondary 5V power delivery is consistent

superjax, 2019-02-11:
I believe this is closed by #335.

closed 2019-02-11 by superjax