rosflight/rosflight_firmware #307: Feature request: Split yaw LPF from the rest so one can tune it's filter independently

korken89, 2018-09-12:
In a system I am currently working on excessive noise in the yaw axis is causing problems, and I have currently solved it by manually setting the yaw rate LPF manually in the code.
To this end, I'd like to request that the yaw rate LPF can be tuned by its own, and not be locked to the other axises.

The rationale is the the yaw axis has generally way less bandwidth than the roll and pitch axises and to reflect this, and since the yaw controller generally has a lot higher gains than the roll and pitch controllers, more aggressive filtering of yaw can be needed.

What are the thoughts from the devs on this?

superjax, 2018-09-13:
That's a good idea.  @dpkoch, what do you think?

It's a relatively straight-forward adjustment.  I'll code up a prototype.

dpkoch, 2018-09-13:
Closed by #308

closed 2018-09-13 by dpkoch