rosflight/rosflight_firmware #311: Per-channel RC override not helpful

drinkdhmo, 2018-10-17:
MAV type: fixedwing

Per-channel RC override does not add any value over complete take-over. In fact, it only sows confusion when trying to tune. It can be difficult to know which channels are being controlled by the companion computer, especially since the status message is binary: RC override is either true or false.

superjax, 2018-10-17:
Thanks for the feedback.

The primary motivation for per-channel override is actually in distinguishing throttle override from attitude.  This is most common in multirotors (for example, I want to quickly "bump" roll of the quadcopter without affecting the thrust).

If bumping the stick in roll causes me to take over throttle control, then it will cause the multirotor to jump, as the throttle being commanded by the rc is unlikely to be close to the throttle command being sent by the computer.  This is less of a problem for the attitude channels.

bsutherland333, 2024-07-10:
This is desired behavior that is being used by our assisted flight mode configurations within ROSplane

closed 2024-07-10 by bsutherland333