rosflight/rosflight_firmware #312: Throttle controlled by companion computer when in Manual mode

drinkdhmo, 2018-10-17:
MAV type: fixedwing

With the auto/manual switch in the manual position, the companion computer was still able to control the throttle.

dpkoch, 2018-10-17:
The attitude and throttle overrides can be mapped to different switches by setting the `RC_ATT_OVRD_CHN` and `RC_THR_OVRD_CHN` parameters. The first thing to check is if your parameter configuration has these set to different channels, and if so set them to the same channel to get the behavior you want. If they were set to different channels and you were still seeing this behavior, then this is a bug.

For reference, this is documented [here](http://docs.rosflight.org/en/latest/user-guide/rc-configuration/#safety-pilot-configuration), and referenced [here](http://docs.rosflight.org/en/latest/user-guide/overview/#arming-errors-failsafe). If we can improve the documentation to make this more obvious, that would be great too.

superjax, 2018-11-08:
I am just following up on this.  Was this actually the issue with the switch being twisted upside down?  @BillThePlatypus

BillThePlatypus, 2018-11-08:
Yes, it was

dpkoch, 2018-11-09:
So this was an issue with the actual transmitter hardware, not the firmware. Can we close this issue then?

BillThePlatypus, 2018-11-09:
Yes, that would be fine

closed 2018-11-09 by dpkoch