rosflight/rosflight_firmware #313: Offboard (computer control) timeout too low, how to change?

korken89, 2018-10-24:
Hi all,

I have been experimenting with some high level controllers that produce roll/pitch/yawrate/trust references to the rosflight system at 20 Hz.
Some times, when the solver takes extra time, the publishing rate is reduced to about 8-9 Hz for a sample or 2 until it goes back to 20 Hz, which causes the rosflight to shutdown the motors (due to the throttle being zero on the RC) and the frame falls and breaks.

It seems like there is an assumption somewhere that publishing rate of the offboard controllers must be above 10 Hz?
If so, how can this be reduced to something like 2 Hz (which is still 0.5s of command holding until the take over happens)? I did not find a setting to update this value.

BR Emil

superjax, 2018-10-26:
Yeah,

It's set here  https://github.com/rosflight/firmware/blob/master/src/command_manager.cpp#L279

That is totally something that could be very easily converted to a parameter.  For now, it's a magic number.

korken89, 2018-10-26:
This does indeed sound sane.
Giving the user the choice makes everyone happy :)

closed 2018-10-26 by superjax

korken89, 2018-10-27:
Thanks for implementing!

PR #316 (2018-10-26): Made offboard timeout value into a parameter
Closes #313