rosflight/rosflight_firmware #315: Switch arming works with non zero throttle

korken89, 2018-10-26:
Hi all,

I have noticed a somewhat dangerous thing when arming using a switch; it allows arming even if the throttle is not close to zero.
If this the intended functionality? I would probably use the same threshold for throttle as the stick arming users to not have an accidental full throttle situation. :)

What are your thoughts on this?

BR Emil

superjax, 2018-10-26:
That's a good point.  We haven't thought about that and there are no conditions to catch it.  I think that's a good idea.

@dpkoch what do you think?

dpkoch, 2018-10-26:
Yeah, that's an excellent catch and just something we missed. Let's implement that for sure.

dpkoch, 2018-10-26:
This check should be done in the state manager rather than relying on the RC module, since we want to disallow arming with high throttle under any circumstance.

We should definitely require that RC throttle be low before arming. However, should we also require that the offboard throttle command (if present) be low as well before arming? The situation I'm worried about is someone having disabled the min throttle override mechanism, sending a high throttle command (especially if they've got some integrator windup while disarmed), then flipping the arm switch and the quadrotor goes shooting off.

To solve that we could either require that both the RC and combined throttle command are always low before arming, or only require the combined throttle command to be low if min throttle is disabled. Thoughts?

dpkoch, 2018-10-26:
Hm, good point. I suppose though that requiring the combined control (which muxes the RC and offboard according to whatever overrides are active) to be low while arming could still work, because then even if they turn off min throttle, they can arm with the override switch on (which will cause the combined control to have the low RC throttle value), then they can switch into offboard control to take whatever command is being sent but their finger is still on the switch so they can immediately override again if it goes crazy.

So in effect, in order to arm this would require that either
  * min throttle is enabled and the RC throttle is low, or
  * RC throttle is low and the RC throttle override switch is on, or
  * there is no RC override, but the offboard throttle is low

Now that I write that, though, I'm wondering if we want to force there to be one additional RC action after arming for the offboard to become active (either raise the throttle if in min throttle mode, or turn off the override switch). This way if something goes weird right after arming, they had to just transition out of a safe override situation and should be able to quickly go back, and it would be harder to casually arm it without being ready to take over again.

So in that case, in order to arm we would require that
  * RC throttle is low
  * Either min throttle is enabled, or the RC override switch is on

How does that sound?

korken89, 2018-10-27:
I think that would work, but I can give more feedback after testing :)
The important part is that it is easy and intuitive, else people start creating workarounds.

PR #317 (2018-11-07): Require low throttle and RC throttle override active to arm
Closes #315

This hasn't been tested yet on hardware, will comment once that's been done
closed 2018-11-07 by superjax