rosflight/rosflight_firmware #321: Documentation: altitude and airspeed control

Georacer, 2018-11-15:
Hello everyone!

This is an issue regarding the documentation (http://docs.rosflight.org/en/latest/). Since it links to this repo for issues, I submitted the issue here.

I have read the entire documentation and the control modes provided by the firmware controller are unclear to me.

Specifically: It is stated that `RC_ATT_MODE` switches between angle and rate control.

Q1: Is this common to both multirotor and fixed-wing modes?
Q2: During multirotor mode, is there any kind of altitude or vertical velocity regulation performed by the firmware? Or is this delegated to the companion computer?
Q3: During fixed-wing mode, is there any kind of airspeed and altitude regulation performed? Or is this delegated to the companion computer?

Especially Q2 and Q3 are quite important to set the expectations for the first test flight and affect safety. I couldn't find any hints answering them, except I grossly overlooked.

Thanks

superjax, 2018-11-15:
Thanks for the feedback - we could certainly improve the documentation.

The vision of ROSflight is to create an autopilot with only the most basic functionality while allowing easy prototyping and expansion on an onboard computer.  Because attitude control of multirotors is impossible without fast inner control loops, attitude estimation and control of multirotors is performed in the firmware.  Fixedwing aircraft are not typically as unstable so pretty much everything is done on the onboard computer and the flight controller acts as a I/O board.

So, specifically:

Q1 - While technically you could command angle and rate mode in a fixedwing, it probably wouldn't perform very well.  There are some assumptions in the nonlinear attitude estimator in the firmware that are broken by a fixedwing on a regular basis, so attitude estimation (and therefore control) doesn't work very well.

Q2 - Altitude control of multirotors is expected to be performed by an onboard computer

Q3 - This is also dedicated to the onboard computer.

For an example fixedwing aircraft autopilot stack, I might point you to https://github.com/byu-magicc/rosplane, which is an implementation of the autopilot described throughout [Small Unmanned Aircraft](https://github.com/byu-magicc/rosplane) by Randy Beard and Tim McLain.

superjax, 2018-11-15:
I might mention as well, we have implemented a pretty good Software-In-The-Loop simulation environment which compiles the firmware directly and runs it exactly as it runs in hardware.  If you wanted to extend ROSflight to enable some of this higher-level control you've talked about, I believe that it wouldn't be too hard develop it in the SIL and I'd be happy to answer any development-related questions to enable this type of work.

Georacer, 2018-11-15:
Thanks for answering my questions. They are crystal clear!

So, this issue is considered answered, yet I would still propose putting some emphasis on what isn't supported on ROSFlight in the documentation.

---

OFFTOPIC:

Additionally, it seems to me that you wouldn't use ROSFlight for attitude estimation in a fixed wing, instead you suggest the ROSPlane sister project. That's fine.

I have also read the Beard/McLain's book and applied it to a RC aircraft with an Arduino, so I know what to expect. It's totally workable.

I'm in the phase where I'm looking to do a fault-tolerant control experiment in a fixed-wing and I dread writing research code in ArduPlane, which I use regularly otherwise as a low-level flight controller.
I'm shopping around for flight controller stacks and your project seems interesting!

superjax, 2018-11-15:
Great!  I hope your research goes well!

closed 2018-11-15 by superjax