rosflight/rosflight_firmware #325: Don't output PWM if invalid mixer

superjax, 2018-11-29:
This has been a problem a couple of times with different users - they plug servos into the flight controller and they burn up because we are outputting motor updates at 490Hz.

I don't think it is wise to change the default PWM output rate to 50Hz because that will make multirotors way less stable.

I propose the following changes:
 1. Do not generate PWM at all if `PARAM_MIXER == INVALID_MIXER`, 
 2. Adding a field to the `mixer_t` struct for the PWM update rate and remove the `MOTOR_PWM_UPDATE_RATE` parameter entirely

This is a relatively small change, but I wanted to get your opinion on the removal of the parameter - are there situations in which people would not want the default 50HZ for fixedwings or 490Hz for multirotors?

drinkdhmo, 2018-11-29:
Some digital servos can handle 300+ Hz update rates. That might be useful for small aerobatic fixed-wing aircraft.

superjax, 2018-11-29:
alright. so here is the new idea for proposition 2,  what if we have a default PWM for each mixer and `MOTOR_PWM_UPDATE_RATE` has a default value of `0`.

If the user wants to specify the pwm update rate other than the default, they can, but if the parameter is `0`, it uses the default for the mixer.

closed 2019-02-11 by superjax