rosflight/rosflight_firmware #329: Arming by sending commands from offboard computer?

abeer-noor, 2018-12-17:
Sorry for posting my query here but the discussion forum is not working. I am unable to post anything there or see past discussions. As a last resort, I am posting my question here.

I am unable to arm the drone while sending commands from the off-board computer. I have tried sending a large value for yaw rate and 0 throttle as this is the default way to arm via RC but this doesnt work.

What is the correct way to arm via sending commands on the /commands topic?

superjax, 2018-12-17:
I'm sorry about the forum, we are having some issues with our webhosting recently and are in the process of moving to a different web hosting provider so we don't keep going down.

Right now, we do not support arming via the computer - http://docs.rosflight.org/en/latest/user-guide/hardware-setup/#rc-transmitter-and-reciever.  This is primarily due to safety concerns, but I could see a case for enabling it.  Could you describe your setup and reason for needing the ability for the computer to arm and disarm the quad?

abeer-noor, 2018-12-18:
I want to perform autonomous waypoint navigation and for that, I wanted to have total control over the drone through my program, which included arming, running on the on-board computer. But you are right about the safety concerns and I understand why you would have made this decision.

But even after arming via RC and then publishing commands on the /command topic, I am still unable to fly the drone in Gazebo simulation. The armed, offboard and rc_override flags are true in the /status message and I can see the commands being published by echoing the /commands topic. But still the drone doesn't fly.

Maybe there is some problem with my parameter configuration. I have attached my parameter configuration file. If you could please take a look at it, it would be super helpful.

[testing_params.zip](https://github.com/rosflight/firmware/files/2689544/testing_params.zip)

dpkoch, 2018-12-18:
If the `rc_override` flag is true, then the flight controller will be ignoring offboard commands. From your parameter file it looks like you're using the default minimum throttle behavior (described [here](http://docs.rosflight.org/en/latest/user-guide/overview/#rc-safety-pilot)). You also have the attitude and throttle override switches mapped to channel four. With those settings, the problem may be one of the following:

  * Your throttle stick is low, and since the flight controller takes the minimum of the RC and offboard throttle commands, it never takes off. If this is the case your options are to either throttle up on the RC transmitter after arming (this is how we typically do it so we can quickly drop the throttle if something goes wrong), or disable the minimum throttle behavior by setting the `MIN_THROTTLE` parameter to 0. If you go the second route, note that you'll need to set your override switch high before you can arm, and then you can switch it off after arming (another safety decision).
  * Your override switch is high, which would cause the flight controller to ignore offboard commands. Try toggling it to see if that helps.

Do either of those suggestions work for you?

abeer-noor, 2018-12-19:
Yes, by disabling both the override switches and setting the `MIN_THROTTLE` parameter to 0, the issue was resolved. Thanks.

But what is the correct procedure to arm the drone first and then controlling it via offboard commands? Because if I arm the drone via RC and then fly using the offboard commands, the `rc_override` flag remains true. I had to kill the `rc_joy` node to disable the `rc_override`. :P

dpkoch, 2019-01-16:
Hi @abeer-noor, sorry for the delay.

Either of the following two procedures are recommended:

  1. *Min throttle method:* (this is what we typically use and is enabled with the default parameters). The `MIN_THROTTLE` parameter is set to 1. We typically also have the throttle and attitude override channels mapped to the same switch, but don't use them (they're enable for safety just in case we need to take complete control). To fly, we start publishing offboard commands, arm the vehicle, then slowly raise the throttle to allow the offboard commands to be used. After we've ensured the offboard commands are doing something reasonable, we raise the throttle all the way so it doesn't limit throttle coming from the offboard setpoints.

  2. *Override method:* Disable min throttle by setting the `MIN_THROTTLE` parameter to 0. Flip the override switch so that throttle and attitude override are active. Set throttle to zero, and arm. Flip the override switch to give control to the offboard setpoints. We don't typically use this method simply because it doesn't give you as much opportunity to check things are behaving how you want before fully giving control to the companion computer, but if you're trying to do an autonomous takeoff sequence it might be useful.

If you're following one of these methods and the `rc_override` flag is still active, one common cause is not raising the throttle stick all the way when in min throttle mode. This is especially relevant when using Xbox controllers since they center at 50% throttle.

Another thing to check is that you're publishing to the `/command` topic at 10Hz or faster, otherwise the commands timeout and the flight controller ignores them.

abeer-noor, 2019-01-17:
Thank you for the detailed explanation. :) I will let you know after trying it.

closed 2019-01-21 by abeer-noor