rosflight/rosflight_firmware #330: GPS link broken

Georacer, 2018-12-31:
Hello everyone,

This is a documentation issue report.
In the hardware page (https://github.com/rosflight/firmware/blob/master/docs/user-guide/hardware-setup.md) you provide a Drotek link for a GPS receiver, which is sadly broken by now.

Perhaps you intend to point to this [this](https://drotek.com/shop/en/u-blox/876-module-gps-ublox-neo-m8n-magnetometre-lis3mdl-pixhawk-3-pro.html) product, but since I'm not sure, I refrain from doing a PR.

Best wishes for a happy new year!

superjax, 2018-12-31:
Fixed.  Thanks!

Also, we typically use [this](https://drotek.com/shop/en/u-blox/883-ublox-neo-m8-gps-module.html) because it doesn't have the external mag, which we haven't found very useful.  Both modules work though.

Georacer, 2018-12-31:
Just out of curiosity, and since I plan to buy an OpenPilot Revo to try ROSFlight out: People say that, with the CC3D software the internal magnetometer is mostly unusable, i.e. very hard to calibrate.
Are you content with the internal magnetometer for use with ROSFlight? Do you actually use it, or you work with a fixed-wing setup?

Thanks!

closed 2018-12-31 by Georacer

superjax, 2018-12-31:
We generally just don't use the mag at all.

We undergo enough accelerated motion to resolve heading from GPS.  However, if you intend on flying straight and level for extended periods, and/or want to estimate wind direction and speed then it will probably be important for you to use an external mag.  (You're right, the one onboard is basically useless).

Currently, the plumbing isn't there to support an external mag on the revo, though it should be pretty easy to add.

superjax, 2018-12-31:
If you wanted, you could actually just buy the acro version of the Revo https://www.readytoflyquads.com/openpilot-cc3d-revolution-acro.  That would save you some money if you're not interested in the mag or the baro.  You could also just populate the barometer yourself if you still wanted that.

Georacer, 2018-12-31:
The hard part is to integrate the sensor measurements to an AHRS filter. I see that you run a Mahoney filter without magnetometer but he actually has a version with magnetometer measurements, just FYI.

Thanks for the board suggestion. I think I'll go with [this one](https://www.banggood.com/OpenPiolot-CC3D-Revolution-Flight-Controller-Oplink-M8N-GPS-Distribution-Board-p-1000084.html?cur_warehouse=CN#jsReviewsWrap). It has a nice bundle, although the GPS doesn't have a compass, I'll have to swap it for another.
It will let me run ArduPilot whenever I want to, it's supposedly supported.

superjax, 2018-12-31:
Sweet.

Yeah, I don't think that fusing the magnetometer in the nonlinear observer would be a great idea on a drone.  I know that paper had success with it, but I couldn't get it to work, so I pulled out the mag measurement model.

Because the mag is so noisy, you have to do a lot of checks to ensure you're not fusing noise.  A Kalman filter is much better suited for mag fusion, because you have the covariance matrix which allows you to do NIS tests (Normalized Innovation Squared).

Georacer, 2018-12-31:
Speaking completely hypothetically, IF I were to port the Madgwick filter witch I have used in the past in an Arduino and which uses magnetometer readings, can you think of a good way to benchmark the performance of the two filters?

superjax, 2018-12-31:
Sorry, I'm not quite sure what you mean.

I think it would be most appropriate if you had a motion capture room and were somehow able to log the attitude estimate of both, then it would be easy to compare.

If you wanted to add magnetometer, I would just copy the block of code starting at L#165 of `estimator.cpp`, replacing all references to gravity with the expected magnetic field and the acceleration reading with the magnetometer sensor.

The most recent magnetometer measurement is accessible as the `turbomath::Vector` `RF_.sensors_.data().mag`

To calibrate the magnetometer, there are 12 parameters, the 3x3 Soft Iron compensation matrix and the 3x1 hard iron compensation constant.  I believe that there is a ROS node which can calibrate the magnetometer from `/mag` topic under `rosrun rosflight calibrate_mag`