rosflight/rosflight_firmware #345: Is it possible to install ROSFlight from source for ROS Melodic?

JonathanPlasse, 2019-06-13:
Hi,

I have a Jetson Nano that I want to use as onboard computer.
The problem is that it run on Ubuntu 18.04.
Consequently, I can only use ROS Melodic.
So, is it possible to install ROSFlight from source for ROS Melodic?

Another question, Will ROSFlight have a ROS Melodic version?

Thanks in advance

superjax, 2019-06-13:
Hi Johnathan,

Yes, it should be very easy to install from source.  I did it yesterday on
a tx2.

Clone the rosflight repo https://github.com/rosflight/rosflight to your
catkin_ws/src folder, install the ros-melodic-eigen-stl-containers and it
should build no problem.  It might throw warnings if you don't have gazebo
installed, which is fine. It's just telling you that it's not building the
simulation features

Yeah, I'm actually not sure why the melodic version isn't on the repos.  I
haven't been a part of the release process, I'll ask Dan.

Hope that helps!

James

JonathanPlasse, 2019-06-14:
Thanks @superjax, I will try it out and keep you updated.

dpkoch, 2019-06-17:
Yeah I've been working on getting it released for Melodic. The issue I'm having is that the Gazebo dependencies seem to be broken for the Debian builds. Hopefully I'll get that figured out soon, then we should be able to release

JonathanPlasse, 2019-06-18:
Good luck

closed 2019-06-25 by superjax

Arsalan66, 2019-11-17:
Sir i have installed ros-melodic-eigen-stl-containers but it's not working :
 sudo apt-get install -y libeigen-stl-containers-dev
[sudo] password for arsalan: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libeigen-stl-containers-dev is already the newest version (0.1.8-1).
The following packages were automatically installed and are no longer required:
  linux-headers-5.0.0-31 linux-headers-5.0.0-31-generic
  linux-image-5.0.0-31-generic linux-modules-5.0.0-31-generic
  linux-modules-extra-5.0.0-31-generic
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 416 not upgraded
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "eigen_stl_containers" with any of the following names:

    eigen_stl_containersConfig.cmake
    eigen_stl_containers-config.cmake

  Add the installation prefix of "eigen_stl_containers" to CMAKE_PREFIX_PATH
  or set "eigen_stl_containers_DIR" to a directory containing one of the
  above files.  If "eigen_stl_containers" provides a separate development
  package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  rosflight/rosflight/CMakeLists.txt:10 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/arsalan/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/arsalan/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:320: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

I would appreciate if someone could help me on this

superjax, 2019-11-17:
Based on the error message, it looks like you need to install `ros-melodic-eigen-stl-containers`

http://wiki.ros.org/eigen_stl_containers