rosflight/rosflight_firmware #346: RC receiver not powering on

vkotaru, 2019-06-18:
Hi, 

I apologize for posting this here, but I was unable to post the question on discuss.rosflight.org. 

I recently got this [board](https://hobbyking.com/en_us/openpilot-cc3d-revolution-revo-32bit-flight-controller-w-integrated-433mhz-oplink.html?___store=en_us) as suggested in [docs.rosflight.org](http://docs.rosflight.org/en/latest/user-guide/hardware-setup/#flight-controller).
However, the [wiring-diagram](http://docs.rosflight.org/en/latest/user-guide/hardware-setup/#wiring-diagram) seems to be using a different board. Can anyone help me out with the right ppm/sbus port to use in the above board (openpilot-cc3d-revolution-revo-32bit-flight-controller)?

I tried the flexi-IO port on the FCU (as shown in this [documentation](https://hyperion-world.com/download_files/manuals/HP-FCCC3DF4R-MAN.pdf)),  however, it doesn't seem to turn on the receiver. 
PS: Receiver was independently tested on a different setup and it works. 

Thank you
Prasanth

superjax, 2019-06-19:
No problem! Sorry about the forum being down.

So, the F4 board doesn't supply power from the USB to the flexi-io port.  You need to somehow supply power to the servo rails.  I use a UBEC or ESC power.

vkotaru, 2019-06-20:
@superjax  Thank you!

closed 2019-06-20 by vkotaru