rosflight/rosflight_firmware #350: OPLink Support

JonathanPlasse, 2019-06-28:
Hello,
I want to use an OpenPilot Revolution as flight controller and I saw that OPLink is integrated in it.
So, is OPLink supported?

superjax, 2019-06-28:
Unfortunately, it's not.

I was originally really excited about the OPLink, because I thought that it was more like the 3DR radios where you basically pass it data and it handles the data handshaking and so on, but it's not.  The driver is responsible for handling all that stuff, so it's actually a lot more development overhead.

I've flown remotely a few times with a 3DR radio as my connection to the base station and that seems to work pretty well, even though I feel silly carrying two radios and not using the built-in OPLink.  I have to configure the 3DR to push up the bandwidth, which reduces my range, but it works okay.

Georacer, 2019-07-08:
Do you recommend plugging in an antenna in the onboard OpLink antenna connector, so as to avoid back-propagation of power and possible damage to the board?

Or do you just leave the connector unplugged?

superjax, 2019-07-08:
I just leave it unplugged.  Maybe that's naive, but I haven't seen any problems so far.

readytoflyquads is selling a replica of the Revo that is missing the OPLink https://www.readytoflyquads.com/openpilot-cc3d-revolution-acro.  It's also missing the baro and mag, but otherwise seems exactly the same.  I don't know if that would be safer or not.

Georacer, 2019-07-09:
That's okay. I guess I'll find out myself soon. Thanks for the reply!

jowaibel, 2019-07-11:
> Unfortunately, it's not.
>
> I was originally really excited about the OPLink, because I thought that it was more like the 3DR radios where you basically pass it data and it handles the data [...]

Hi, how do you configure the 3DR radio then? Is there any tutorial on how to use it with rosflight? What port do you connect it to and what software do you use to show the telemetry data on your base station? Thanks!

closed 2023-03-25 by JonathanPlasse