rosflight/rosflight_firmware #359: IMU timing problem

prasgane, 2019-07-30:
Hello,
I flashed the latest version of ROSFlight onto a Flip32 board and I subscribe to the IMU topic (imu/data) and it looks like the accel values are reporting nan. 

Upon closer inspection it looks like the dt used in [get_filtered_IMU](https://github.com/rosflight/firmware/blob/master/src/sensors.cpp#L345) function is zeros. This could be because the function could be called twice before an IMU transaction occurs. We get a zero dt (and hence a nan accel) every ~5 secs. 

I have verified this on two separate boards.  This could be happening with the gyro too but we have not verified it. This looks like a big error and we hope you could fix this soon.

dpkoch, 2019-08-05:
Thanks for reporting this. We'll work on adding a check to make sure this doesn't happen. In addition:

  * We highly recommend switching to an F4 flight controller. The F1 is no longer actively supported
  * Are you streaming IMU at 1000Hz? This timing issue should only happen if you're streaming at full rate. For most applications that's overkill, so you could reduce your IMU streaming rate to something like 250Hz

prasgane, 2019-08-05:
Thank you for your response. 

We were actually streaming the IMU at 500Hz. I dont think the IMU rate is a problem because when we tried running our application with the old firmware (1.2.0) we did not face this issues (even when streaming at 500Hz). So I think there might be problem when the firmware was update.

Additionally, we are also experience problems when closing the loop by publishing to the /command topic. The new firmware causes an erratic behavior for the same code that ran smoothly on 1.2.0.   

I am not sure if these two issues are related.

dpkoch, 2019-08-06:
Interesting. What is the erratic behavior you are seeing?

prasgane, 2019-08-07:
Erratically meaning, the response is not smooth and the quad flip over. 

The quad flies "fine" (the PID gains have to be tuned) when flying manually using an RC controller.

prasgane, 2019-08-08:
I can try getting bag file and a video if it helps.

dpkoch, 2019-08-08:
That would be helpful. Do you also happen to have an F4 board available? I'm curious if the issues you're seeing happen on both boards, or only on the F1.

bsutherland333, 2023-10-04:
I'm not sure if this is still an issue, but it's probably worth checking before closing.

bsutherland333, 2024-05-31:
We haven't seen this issue with our current hardware (H7 based boards) so I'm going to close this issue.