rosflight/rosflight_firmware #360: Instant flip on takeoff

emiilbjorklund, 2019-07-30:
Hi,
First of all, great work developing this product.

Im experiencing some issues when trying to takeoff with my drone using rosflight. The hardware that is used as an FC is a Naze32 F1 and it works "okay" when the Naze32 firmware is loaded.

Im just wondering if there is a way to see the IMU alignment in rosflight because my guess is that maybe something is off there. And also, the default PID values, is it possible to test fly on those or does they need tuning before first flight?

I have set the correct 'MIXER' (2) and also checked the RC output against the website.

Am I missing something? 

Best regards,
Sweden

dpkoch, 2019-08-01:
Hi @emiilbjorklund,

Probably the easiest way to check the IMU alignment is to `rostopic echo /imu/data/linear_acceleration` (or use rqt_plot) and check that the values are sensible. You should see -9.8 in the z-axis when the vehicle is flat, +9.8 in the x-axis when you pitch up 90 degrees (nose up), and +9.8 in the y-axis when you roll left 90 degrees.

If those are wrong, check the values of the `FC_ROLL`, `FC_PITCH`, and `FC_YAW` parameters, which describe the orientation of the flight controller with respect to the body. Make sure they're either zero or the correct values for your setup.

If those are all good and the alignment is still wrong, we'll have to do some more digging. We have had issues with faulty IMUs on some cheaper Naze32 clones. Typically that has manifested as the magnitude being off by a factor of 2 (reporting 4.9 instead of 9.8).

Most quads should fly reasonably well with the default PID values.

prothen, 2021-03-21:
I am experiencing the same behaviour with a Xrotor nano F4, is there something that you recommend to look into as a start?

closed 2023-10-04 by bsutherland333