rosflight/rosflight_firmware #362: Documentation and Discuss Problems

DaneSlattery, 2019-08-20:
Hi, I tried to post this onto: https://discuss.rosflight.org/post/discussion

When trying to connect to https://docs.rosflight.org/ , I get: 

Your connection is not private
Attackers might be trying to steal your information from docs.rosflight.org (for example, passwords, messages, or credit cards). Learn more
NET::ERR_CERT_COMMON_NAME_INVALID

AND when trying to post on the forum I get: 

{ "Code": 403, "Exception": "Invalid CSRF token. Please try again.", "Class": "Gdn_UserException" }

Something seems terribly wrong.

DaneSlattery, 2019-08-20:
More information would probably help:

I'm on Windows 10, using Google Chrome, but this certificate error occurs on my Phone, on my friends MacBook, which leads me to believe this affects more than just me.

DaneSlattery, 2019-08-20:
The Docs seem to have magically fixed themselves right now, but this error is recurring every now and then. 

The forum CSRF Token Error persists!

dpkoch, 2019-08-22:
Hi @DaneSlattery, sorry for the inconvenience.

We recently moved where we host docs.rosflight.org, so that was most likely a caching issue that your browser eventually figured out.

We've been having troubles with the forum, and are in search of a new solution. That's been progressing slower than might be hoped, but hopefully we'll get that fixed in the near future. Sorry about that.

DaneSlattery, 2019-08-25:
Hi @dpkoch , thank you for letting me know, at least you were aware of the issue, and now others will be too!

dpkoch, 2019-08-29:
Hi @DaneSlattery, thanks for the helpful info!

It looks like the documentation certificate error is widespread enough that my initial thought that it's a caching error may be wrong. It looks like there is something funny with how Github pages is setting up its HTTPS/certificate? I'll look into this more.

rafaelmaeuer, 2019-10-12:
Hi, when I try to register at https://discuss.rosflight.org I get a page not found error:
[image]

superjax, 2019-10-14:
@dpkoch, do we have a direction on this?  I can't remember what the plan was to fix this.

plusk01, 2019-10-21:
Personally, I like the low-commitment and visibility of a forum over slack.

@dpkoch, I may be able to dust off some of my web dev skills to help with this.

I am able to login, but when I try to create a test post, I get:

[image]

superjax, 2019-10-22:
I'm all for the slack channel.  I know that the conversations get lost after a while, but a 10,000 message limit is a lot better than a broken system.

Unfortunately, I'm also not the slack owner, that's @dpkoch's wheelhouse.  I think we'll have to wait for him to get back.

dpkoch, 2019-10-31:
The new forum is up and running! I've setup a discourse instance on a virtual private server, so now we have a nice modern and clean solution. The site is live, at the same URL (https://discuss.rosflight.org).

@superjax, @plusk01, could you check it out and let me know of any issues? I've got some weird security issues cached on my machine from the setup process, so could you also let me know if HTTPS is working correctly and if Chrome/Firefox/whatever reports the site as secure?

I'm also hoping to eventually move our documentation hosting to this server to try to avoid the certificate issues we've been getting with GitHub pages.

superjax, 2019-10-31:
It's working for me!  Thanks.  Only firefox here, but it's working :)

rafaelmaeuer, 2019-10-31:
@dpkoch awesome work - thank you very much!

dpkoch, 2019-11-13:
Current status:
  - All hosting has been moved to a new VPS
  - New Discourse forum is online (unfortunately I was unable to migrate the old forum contents)
  - Documentation hosting has been moved to new server; this should eliminate the certificate issues we were seeing with GitHub pages
  - A webhook has been configured on the new server to automatically deploy new documentation on push events to master branch

The last thing is to remove the now unused GitHub pages deploy step, which is done in #367 and which will close this issue. All problems discussed in this thread should now be resolved.

superjax, 2019-11-14:
Thanks @dpkoch,  this is huge.  

I should mention, `rosflight.org` is down for me right now.  It was working for me eariler though.

PR #367 (2019-11-14): Remove mkdocs github pages deploy from travis script
This pull request removes the GitHub pages deploy step in our Travis configuration. Since I've moved documentation hosting to our own server to avoid the certificate issues we're having, that step is no longer needed.

Along with previous changes, this closes #362
closed 2019-11-14 by dpkoch