rosflight/rosflight_firmware #364: Which compatible STM32F4x5 board to buy?

rafaelmaeuer, 2019-10-12:
As the discussion board doesn't allow user registrations at the moment (see [362#](https://github.com/rosflight/firmware/issues/362#issuecomment-541307026) ) I need to ask the question here as there is no other channel available at the moment.

As the CC3D Revo is really hard to order today (hobbyking, etc. discontinued) and the only available options come from china, I need to ask which STM32F4x5 compatible board should I buy to use ROSflight? The latest releases only include naze32.hex so is F4 support available already?

Buying an old F1 controller seems to be the wrong invest in a time where F7 controllers become popular. There are good F3 options available but as of [299#](https://github.com/rosflight/firmware/issues/299#issuecomment-391193573) it seems they won't be supported right?

maxmine11, 2019-10-12:
That is exactly what I'm wondering! I've been looking everywhere for the rosflight-F4.bin listed on the rosflight website but no luck yet.

superjax, 2019-10-14:
Hey guys,

F4 support is stable except for some of the I2C peripherals.  We think we have squashed the bugs there, but we are still doing some testing.  @dpkoch is a lot closer to this development so he can speak better to that.

Last I knew, the I2C stuff was limited to weird bugs when using the airspeed and sonar sensor.  If you're flying a multirotor, then the F4 board is definitely stable, and has been for several months.

I have been using the STM32F405-based boards.  Specifically, anything directly derived from the OpenPilot Revo F4.  Like [this](https://www.readytoflyquads.com/openpilot-cc3d-revolution-acro) one

Because of these bugs (which have been surprisingly difficult to squash completely, as they are really intermittent and difficult to reproduce) we haven't officially released the F4 firmware.  Therefore, you just have to build it from source.

rafaelmaeuer, 2019-10-19:
As the Revo Acro is only available in USA und the normal Revo in China only, I tried to go with a Naze32 first, as its good available for me. When I received the FC it turned out that its a Flip32, but as described in the docs, it should be compatible though... 

When flashing the firmware with CleanFlight failed (stuck on blue led) I needed to recover the FC with STM32 Flash loader demonstrator and then flashed ROSflight firmware (release 1.2.0) with this tool.

Now the FC shows blue solid LED in combination with red blinking LED. Is that the expected behavior? How can I check the firmware is correctly running on the board? As red is usually an indicator for errors, I'm not so convinced the firmware is correctly working.

rafaelmaeuer, 2019-10-21:
Okay I got an instance of Rosflight running on a companion computer. When connecting to the FC with `rosrun rosflight rosflight_io _port:=/dev/ttyUSB0 output="screen"` it connects and tells that firmware is v1.1.0 (strange as I flashed v1.2.0 on it). Nevertheless, red blinking seems to be okay and not a fatal state. But there are some critical errors afterwards:
[image]
If I run `rostopic echo /rc_raw` I only get zeros for every channel. My RC doesn't seem to be recognised correctly... The receiver shows connection with the RC, and I set the mode to `PPM` on the RC and the FC. Any ideas?

superjax, 2019-10-21:
Make sure that the `RC_TYPE` parameter is correct

https://docs.rosflight.org/user-guide/parameter-configuration/#description-of-all-parameters

rafaelmaeuer, 2019-10-22:
This is exactly what I did, if I call `rosservice call /param_get RC_TYPE` now it outputs 
```
exists: True
value: 0.0
```
which means PPM-mode right? Transmitter-Output is set to PPM too.

I use a FS-IA6B Receiver with a FS-TM10 Transmitter and followed this tutorial which is straight forward: https://www.youtube.com/watch?v=yN00UdPQdRc&t=3s

superjax, 2019-10-22:
Cool.  After setting `RC_TYPE` did you reboot the FC?

rafaelmaeuer, 2019-10-22:
Yes, after flashing the firmware again the green LED started to display solid state. I checked raw input and get the Transmitter values. I then changed RC_TYPE to 0 and solid green disappeared. Seems that RC_TYPE 1 seems to work for PPM although it should be for S-Bus...?

ChrisGermany, 2019-12-16:
Hi I wanted to ask if you got the Flip32 F4 or the Flip32 F1? I tried to find a revo F4, but the hobbyking FC has been discontinued. I am looking at an alternative F4 board that I could integrate. I have an Omnibus F4, but that does not seem to be supported yet.

rafaelmaeuer, 2019-12-16:
Hey I have the F1 version, but I am looking for a F4 version as well. The CC3D Revo is pretty hard to get, you can still find some boards on eBay... I agree, it would be nice to have a buyable alternative for a compatible F4-Controller!

rafaelmaeuer, 2020-01-23:
So after another round of research I have found another STM32F405 based FC called [REVO F4 Flight Controller](http://copterjungle.de/f4-revo-stm32f405-flight-controller/). When comparing with available boards on readytoflyquads, it seems that its pretty much the same as the [FLIP32 F4 (REVO PIN HEADER EDITION) V2](https://www.readytoflyquads.com/flip32-f4). As this FC is available at [ebay](https://www.ebay.de/itm/Betaflight-CC3D-REVO-F4-STM32F405-Flight-Controller-With-Vbat-Buzzer/352671711059) shipping from Europe I will give this one a try...

vkotaru, 2021-07-24:
@rafaelmaeuer Have you been able to fix the issue?

rafaelmaeuer, 2021-07-24:
I did use the REVO F4 with rosflight in [this](https://github.com/rafaelmaeuer/Autonomous-Indoor-Drone) project, but at the end the move from simulation to hardware wasn't successful due to lack of time and some stabilisation issues. Until then I didn't gave this FC another try, but I will in the future and then share my experiences here...

bsutherland333, 2023-10-04:
Current plans going forward are to use H7 controllers.

closed 2023-10-04 by bsutherland333

xRJZ, 2024-09-25:
> Current plans going forward are to use H7 controllers.

Any more updates on this? Is the cube orange available?

bsutherland333, 2024-09-25:
We've actually decided to go 3DR's Pixracer Pro due to a simpler board configuration. We've done most of the work required for adding support for the board, and are just doing some final testing. We're done a few flights with it now and are just working out some lingering bugs. 

https://store.3dr.com/pixracer-pro/

@iandareid @JMoore5353 what's the status on Pixracer Pro documentation?