rosflight/rosflight_firmware #372: Changing MIXER causes RC values to read incorrectly

jbwillis, 2020-01-31:
Using the current `master` branch with the default parameter configuration, I am having the `/rc_raw` values jump to being between `985-989` after changing the MIXER parameter with the sticks centered and throttle down. I am using an frsky V8R4 receiver connected to an OpenPilot Revo via PPM. 

Talking with @dpkoch and @BillThePlatypus we think it is likely due to the `pwm_init` function being called every time the MIXER parameter gets changed. This could be causing changes to the timer configuration for the timer used for reading the PPM.

One note, this caused the throttle to jump to 100% and the arm switch value to jump to 100% and hang regardless of the values on the controller. Luckily the vehicle I was working with wasn't armed when the MIXER switch was made, otherwise it would have been entirely uncontrollable.

BillThePlatypus, 2020-01-31:
When you say it wasn't armed, do you mean that there was some sort of physical switch disconnecting the motors? Could this cause an disarmed plane to arm itself?

jbwillis, 2020-02-04:
By not armed, I mean that rosflight was not in the armed state.

BillThePlatypus, 2020-02-05:
That is a safety issue. You don't want a disarmed aircraft arming itself, especially when it can simultaneously misread the throttle command.

BillThePlatypus, 2020-02-06:
I've replicated the issue. I'm surprised we haven't seen this yet.

jbwillis, 2020-02-06:
It might be worth working back through the history of rosflight and
airbourne to see if there’s a change that broke it.

PR #376 (2020-02-07): New airbourne to prevent PPM RC from flipping out on mixer change
Fixes #372 
This change to Airbourne disables PWM output 11, which conflicted with RC over PPM. This also prevents a potential safety issue, as the conflict could cause the flight controller to arm itself while changing parameters.
closed 2020-02-07 by dpkoch