rosflight/rosflight_firmware #396: Failed to open serial port

roboticsai, 2020-05-20:
I'm trying to flash rosflight on px4pilot following [this](https://docs.rosflight.org/user-guide/flight-controller-setup/) tutorial. But after i load the local rosflight_NAZE_Release.hex file and Short the boot pins on your flight controller, unplug it, and and plug back in.
. i get the above error.

superjax, 2020-05-20:
which error?

superjax, 2020-05-20:
also, what is the CPU on px4pilot?

prasgane, 2020-09-10:
@superjax I had the same problem @roboticsai is having. I was trying to flash a Flip32 F4 board with ROSFlight v1.3.0 (downloaded the .hex file from [here](https://github.com/rosflight/firmware/releases/tag/v1.3.0)). I used CleanFlight to flash the firmware and saw a Programming: SUCCESSFUL message but after flashing the firmware, the blue light would stop blinking and only the green flight would stay on. When you try to run the rosflight node, it would say something like failed to open serial port. 

I fixed this by building the firmware from source as mentioned [here](https://github.com/rosflight/firmware/blob/master/docs/developer-guide/building-flashing.md) and flashing the hex generated from this using CleanFlight (not the terminal). I would check if the .hex file is valid in the Releases page is still valid and if it works.

Prashant

superjax, 2020-09-11:
Yeah, mixing up F1(NAZE) and F4(REVO) firmware definitely won't work. It'll say the flash was successful, but it won't run.

superjax, 2020-09-11:
Another quick diagnostic is to watch `dmesg` while plugging in the flight controller.  You should see messages as the device registers itself with the usb drivers.

prasgane, 2020-09-11:
Do you think it would be possible to upload the F4 firmware to the [Release page](https://github.com/rosflight/firmware/releases/tag/v1.3.0)? For someone following the instructions on the [Flight Controller Setup](https://docs.rosflight.org/user-guide/flight-controller-setup/) page, its not really clear the REVO means F4 board and NAZE means F1. Additionally, since you are encouraging users to move to the F4 boards, it would make more sense to have the REVO version on that page than the NAZE version.

bsutherland333, 2023-10-04:
I believe this has been done.

closed 2023-10-04 by bsutherland333