rosflight/rosflight_firmware #397: Support for servos

malioni, 2020-05-20:
We are planning on using RosFlight with the S900 quad, which has 6 rotors and also a servo that lifts/lowers the arms when needed. From what I understand, there is currently no support for a servo to be added to the F4 with RosFlight. The main issue comes from servos needing different update rates than the rotors, and currently RosFlight only provides one update rate. Can there be added support for different update rates out of certain PWM pins so that a servo could be added and controlled through the flight controller?

superjax, 2020-05-20:
Cool!

engband, 2020-05-21:
repeating here from the slack channel, so everything is together:

The S900 has 6 rotors plus landing gear.
The S1000 has 8 rotors plus landing gear

I am not sure at the moment if the landing gear requires two separate signals for the two separate servos, or if the Power Distribution Board has a muxer for the signal. I will edit this comment as soon as I find out. Consequence is, they might need two servo pins with the same signal on them, vs one servo pin.

If the flex-io port is used for the servo lines (this sounds not likely, currently?) then perhaps we can mux one flex-io pin on the "squid" board, and place on the "squid" an amplifier and signal split out to the two landing gear, if that makes sense. It's probably possible, but more hassle and complex hard-ware-wise.

malioni, 2020-05-21:
If I understand this correctly and looking at the schematic provided by James, most of the PWM pins on the Flex-IO port are governed by Timer 8, except for pins PB14 and PB15 (pins 5 and 6 on the port), which are governed by Timer 12. PB14 is usually used for the PPM input from the receiver, but, in my opinion, PB15 could be used for the servos, it still can provide PWM output. Timer 12 is not used on any other pins from what I can tell, and the last two rotors for S1000 could be added to pins PC6 and PC7 (pins 7 and 8 on the port). I might be wrong, but it seems like PB15 could be reserved for this kind of servos if the Timer 12 update rate could be made lower than the other timers.

bsutherland333, 2023-10-04:
@engband Were you ever able to figure this out? I think I remember your S900 and S1000 being able to do this when they were running ROSflight.

bsutherland333, 2024-07-10:
Implemented with with pull request #426

closed 2024-07-10 by bsutherland333