rosflight/rosflight_firmware #398: Arming from companion computer

richiross, 2021-03-20:
Hi everybody
is there a way to arm the drone from the companion computer, thus overriding the rc?

superjax, 2021-03-23:
Not officially, no.    We didn't add this because it's not particularly safe.

However.  If you wanted to, you can make this happen.

Arming the flight controller is done by calling 

`RF_.state_manager_.set_event(StateManager::EVENT_REQUEST_ARM);`
(for example, when arming with RC)
https://github.com/rosflight/firmware/blob/master/src/rc.cpp#L227

Adding a new case to 
https://github.com/rosflight/firmware/blob/master/src/comm_manager.cpp#L279
that dispatches out to this event will allow you to arm from the companion computer.  It's basically just some plumbing  (You'd have to add a new service to `rosflight_io`, and an enum to the mavlink protocol.

richiross, 2021-03-23:
Got it, thanks for your answer!

richiross, 2021-06-29:
May i ask which file refers to the mavlink protocol ?
Thanks in advance

closed 2021-08-30 by richiross