rosflight/rosflight_firmware #416: Update mixing system to be more flexible

JMoore5353, 2024-01-25:
In order to set up a mixer for a custom frame (v-tail for example), a new mixer is required. In the current implementation, this is done by creating the new mixer and recompiling / flashing the firmware. 

In addition, sometimes the user needs to reverse the direction of a servo (change the sign on the corresponding mixer). The current implementation supports reversing for aileron, elevator, and rudder. In the case of a v-tail, however, sometimes individual servos might need to be reversed.

Implementing a way to adjust mixer matrices dynamically (without having to recompile and reflash) could be useful.

bsutherland333, 2024-01-25:
For our v2 release, we'll need to add a mixing for a v-tail style plane since our test platform (the RMRC Anaconda) is a v-tail plane.

bsutherland333, 2024-02-26:
V-tail mixing has been added, but implementing a more flexible mixing system entirely may still be a useful improvement.

closed 2025-01-22 by JMoore5353