rosflight/rosflight_firmware #422: Arm Switch State Machine

bsutherland333, 2024-02-26:
During a flight test, we found that we had issues trying to get the firmware to arm, as it would give an error that throttle override needed to be active to arm regardless of switch position. This was only an issue when arm was set to channel 5 & and both overrides were set to channel 6 (0 indexed). Making sure the unit tests include non-typical switch conditions, and fix any issues that those might bring up. 

Adding some additional documentation on this may also be useful.

bsutherland333, 2024-05-31:
I've realized that there is a parameter that sets the maximum number of channel, which defaults to 6 channels. This may explain the issue we're seeing but it would be good to confirm this. We may also consider setting the default of that to the max number of channels, which is 8. Or remove the parameter altogether and let it be a fixed 8 channels as I'm not sure it would ever be beneficial to restrict the number of channels.

melayton, 2024-05-31:
That would make sense! I can test to make sure the default channel number is the issue and then remove the parameter if that's the case.