rosflight/rosflight_firmware #432: New mixing system

JMoore5353, 2024-08-02:
In the current system, the controller outputs (in Nm or N) are mixed using the mixing matrices defined in `mixer.h`.

JMoore5353, 2024-08-12:
@bsutherland333 @iandareid I have some thoughts I'd like your input on.

I implemented a mixing system that dynamically takes in the number of motors and the motor parameters to create the mixing matrix, using the Eigen library. It works well in sim. However, it dynamically allocates memory on the heap to do this (since the size of the matrices are defined at runtime). We can't use (or would strongly prefer not to use) a heap on the microcontroller, which makes this challenging.

A couple ideas:
- Precompute and hard code some mixing matrices for common frame types. 
    - Pros:
        - Doesn't use Eigen / dynamically allocated memory
        - Doesn't need to compute the mixing matrix on the microcontroller
    - Cons: 
        - Can't adapt for different motor positions (i.e., if we have  a big quad frame vs a small quad frame -- this would change the mixing matrix, but we wouldn't be able to update it unless we recompile).
        - Restricted to frames / configurations that we decide, needing a recompile for other configurations.
- Compute the mixing matrix in `rosflight_io` node and send each entry in the matrix as a parameter to the firmware
    - Pros: 
        - No dynamic memory allocation / extra computation in the firmware.
        - Completely flexible since we can do everything dynamically on the computer
        - Can take advantage of ROS2 array parameters to pass in motor locations (instead of passing them one by one to the firmware)
     - Cons: 
         - Have to add >=40 mixing matrix parameters (4 x `NUM_MIXER_OUTPUTS`)  to the firmware param list

I think my vote would be to do both of the above, meaning we have precomputed, hard coded matrices for specific configurations (like it is now), and then add a `custom_mixer` option that is loaded from params from `rosflight_io`.

What are your thoughts?

JMoore5353, 2024-08-13:
Update: I can use Eigen statically by computing the pseudoinverse using the SVD. I just allocate matrices of the max size and do it that way. It didn't work before because I was trying to use the built in method `pseudoInverse()`, which computes the pseudoinverse in a different method. I couldn't figure out how to get that other method to allocate memory statically. This SVD method should work well, however.

I think the best course of action then is to keep the computation in the firmware. I will still precompute some mixing matrices for some standard frames, but also give the option for a custom matrix with the values specific to the frame (which should be the recommended route).

bsutherland333, 2024-08-13:
I like the idea of keeping the mixing on the firmware. The less responsibilities we assign to rosflight_io the better, since it's supposed to be an io node only. I'm glad you were able to figure out how to compute that without needing heap memory. 

How does the user set custom matrices? It would be nice to have this ability for fixedwings as well. STM processors have PWM outputs in common groups, where the output rate for each group must be the same, making mixing more complicated there as well. Having a highly flexible and easy to configure method of changing channels for the mixing will be very useful.

bsutherland333, 2024-08-15:
@avtoku has some thoughts about this as well, in regards to the fxyz channel input.

JMoore5353, 2024-08-16:
Sounds good. There are times we will need more than just the fxyz values, i.e., a quadplane, where we have an fx and a fz, in addition to the xyz moments.

JMoore5353, 2024-08-20:
It would be good to move the mixer to the Jetson for flexibility.

Thoughts if we mixed on the Jetson:
- For a dynamically changing matrix, we might want to send mixed commands... Otherwise we'd suffer from latency as we sent 60 parameters to the firmware at each timestep
- We'll add ~60 more parameters to firmware
- Would have to add mixer functionality to rosflight_io (or another node?)

bsutherland333, 2024-08-20:
What situations would require a dynamic matrix? And if we're concerned about passing over that much data as parameters, we could add support for matrix or array parameters to the io. The most of our mavlink messages are already custom, so needing to create another custom one wouldn't be a terrible thing. 

I think I'm worried about how we could keep the minimum functionality required for a safety pilot to fly an aircraft on the firmware if the mixing isn't happening on there.

JMoore5353, 2024-08-20:
Airframes like a tricopter, tiltrotor, etc., where the motors are physically moving would require a matrix that changes based on the angle of the motors. 

Depending on how it gets implemented, we might want to add a mavlink message that supports vectors.

The safety pilot concern is something I've been concerned about as well. I am going to look into how PX4 and other platforms mix the tricopter or tilt rotor frames to understand a bit more what they are doing. It might be possible to make some simplifying assumptions for an RC pilot to be able to calculate a mixer (I'm thinking we could assume all yaw comes from the rear motor being rotated in a tricopter, for example).

closed 2025-01-22 by JMoore5353